luksal/ESP32-DMX #4: Case missmatch in SimpleRX.ino

macdroid53, 2019-12-11:
Disclaimer: this could be an issue with my test bed. I'm using VSCode & platformio.
The code in SimpleRX:

`if(DMX_healthy())
...
...
    Serial.print(DMX_read(1));`

But, the definition in .cpp & .h:

`static uint8_t Read(uint16_t channel);  

        static bool IsHealthy();  `

I changed SimpleRX to:
`if(DMX::IsHealthy())
...
...
Serial.print(DMX::Read(1));
`

And it appears to work.

luksal, 2019-12-11:
I'm sorry, there was a type in the example. It is now fixed

closed 2019-12-11 by luksal

macdroid53, 2019-12-12:
Ok, I figured as much... 
Just curious (and showing my limited C/C++ knowledge probably...)

What is the define used for in the .h file? (I understand the ifndef/define, but what is the DMX_h?)

`#ifndef DMX_h
#define DMX_h`

luksal, 2019-12-12:
That is called an "include guard", that's a common pattern to prevent that a header file is included multiple times.
More information: https://en.wikipedia.org/wiki/Include_guard

macdroid53, 2019-12-12:
Duh...looked at the link and connected the #endif to #ifndef. Must have jogged some cobwebs in my brain.

Thanks!

commit 4c2050ed (2019-12-11): Fix example

closes #4