luksal/ESP32-DMX #5: Are other libraries required?

greiginsydney, 2019-12-12:
Hey @luksal, I was really pleased to see you've published this, but I can't get it to compile.

I get stopped at line 1: "dmx.h: No such file or directory" - but I've copied your dmx.h and dmx.cpp into the same folder as my sketch, and that approach has worked for me on projects in the past.

I've created folders for 'freertos' and 'driver' and created the appropriate files from GitHub. 

If I change #include <dmx.h> to "dmx.h" at the top of my sketch the error changes to "DMX.h: No such file or directory", referencing the upper-cased #include at the top of dmx.cpp.

Are there other libraries that this is looking for, and which I might not have installed?

Many thanks.

- G.

luksal, 2019-12-12:
Hello @greiginsydney, what development environment and hardware platform are you using?

greiginsydney, 2019-12-12:
Arduino IDE & a SparkFun Thing Plus - ESP32 WROOM, driving the SparkFun ESP32 Thing Plus DMX to LED Shield.

luksal, 2019-12-12:
You don't need to download the freertos libraries, because they are part of the ESP-IDF framework wich runs in the background of the Arduino core.
Are you running the latest Arduino IDE? Do you have a recent version of the ESP32 library installed via the BoardManager?

luksal, 2019-12-12:
I'm closing this issues, because it's seems that there are problems with your development environment.
Please feel free to ask question in this issues if you need more help getting the code running

closed 2019-12-12 by luksal

greiginsydney, 2019-12-12:
Sorted it! I'd created dmx.h and dmx.cpp in my sketch's folder, but turns out I need to treat it like its a library.

I created a new folder called ESP32-DMX-RX under %UserProfile%\Documents\Arduino\libraries\ and moved the files there - and now it compiles OK!

(I also deleted the other bits, as you indicated earlier).

Thanks for this.

luksal, 2019-12-12:
Great to hear that i works now!