luksal/ESP32-DMX #11: Can't compile lib

IchKG, 2020-02-24:
Hi 
I'm trying to receive DMX and send 9 channels through ESP-NOW.
I bought a SparkFun DMX Thing an connected it to a ESP32 Thing (not plus)
First I tried the SparkFun DMX Lib but got the same issues as several other with shifted data or complete wrong data.
Because of the issues forum in that lib I found luksal's and want to give it a try but I can't even compile it. I downloaded it from github as zip and used the import lib by zip function in arduino IDE (18.12).
When I try to compile the SimpleRX example I get "error: 'uint8_t' does not name a type" and several others. What am I doing wrong?
Complete output:
`In file included from C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:1:0:

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:9:16: error: 'uint8_t' does not name a type

         static uint8_t Read(uint16_t channel);              // returns the dmx value for the givven address (values from 1 to 512)

                ^

In file included from C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:1:0:

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:11:16: error: 'uint8_t' does not name a type

         static uint8_t IsHealthy();                            // returns true, when a valid DMX signal was received within the last 500ms

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:16:16: error: 'QueueHandle_t' does not name a type

         static QueueHandle_t dmx_rx_queue;                  // queue for uart rx events

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:18:16: error: 'SemaphoreHandle_t' does not name a type

         static SemaphoreHandle_t sync_dmx;                  // semaphore for syncronising access to dmx array

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:20:16: error: 'uint8_t' does not name a type

         static uint8_t dmx_state;                           // status, in which recevied state we are

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:22:16: error: 'uint16_t' does not name a type

         static uint16_t current_rx_addr;                    // last received dmx channel

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src/dmx.h:26:16: error: 'uint8_t' does not name a type

         static uint8_t dmx_data[513];                       // stores the received dmx data

                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:19:20: error: 'void* DMX::dmx_rx_queue' is not a static data member of 'class DMX'

 QueueHandle_t DMX::dmx_rx_queue;

                    ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:21:24: error: 'void* DMX::sync_dmx' is not a static data member of 'class DMX'

 SemaphoreHandle_t DMX::sync_dmx;

                        ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:23:14: error: 'uint8_t DMX::dmx_state' is not a static data member of 'class DMX'

 uint8_t DMX::dmx_state = DMX_IDLE;

              ^
[... 109 lines cut ...]
 #define xSemaphoreTake( xSemaphore, xBlockTime )  xQueueGenericReceive( ( QueueHandle_t ) ( xSemaphore ), NULL, ( xBlockTime ), pdFALSE )

                                                                                             ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:132:32: error: 'current_rx_addr' was not declared in this scope

                             if(current_rx_addr < 513)

                                ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:134:33: error: 'dmx_data' was not declared in this scope

                                 dmx_data[current_rx_addr++] = dtmp[i];

                                 ^

C:\Users\markus.langecker\Documents\Arduino\libraries\ESP32DMXRX\src\dmx.cpp:145:21: error: 'dmx_state' was not declared in this scope

                     dmx_state = DMX_BREAK;

                     ^

exit status 1
Error compiling for board ESP32 Dev Module.
`

luksal, 2020-02-24:
Looks like there are some problems regarding your build environment. You can try adding  `#include <stdint.h>` on top of the `dmx.cpp` file inside the src folder.

IchKG, 2020-02-25:
> Looks like there are some problems regarding your build environment. You can try adding `#include <stdint.h>` on top of the `dmx.cpp` file inside the src folder.

Yes that did the trick, I added 
 `#include <stdint.h>`
 `#include <stdstring.h> // for bzero function`
on top of dmx.cpp and moved the include of dmx.h down to be the last one because dmx.h needed queue.h to be included first.
Now it is working fine and its the first library that worked and got all 512 Channels right and I've tested a few. Thank's @luksal for your lib.

closed 2020-02-25 by IchKG

sasodoma, 2020-04-17:
Hello, I would like to ask whether this is a problem with this library, a problem with my IDE or is it like this by design and won't be changed. I ran across the same problem today and it seems strange to me that a fix is known but not implemented in the code. The only two reasons I could think of is that something is wrong with my Arduino installation, or that the Arduino IDE is simply incapable of correctly compiling this without including some extra stuff. Is this problem something that will be fixed or should I fork this library and keep a separate version of it that works with the Arduino IDE?

luksal, 2020-04-18:
I'm not using the library inside the Arduino environment. You are free to create a pull request, so that I can merge the fixes that are needed for Arduino.