luksal/ESP32-DMX #12: There’s sometimes an extra “0” inside the stream shifting all values to the next channel.

satoer, 2020-03-03:
Hi, first of all, thank you for the library. From all the DMX library’s for the ESP32 this one seems to work the best. Sadly in my case, not perfect.

I’m using a RS485 module with a ESP32 (the DMX cable is terminated). I want to read the incoming DMX stream, and this seems to work almost fine. The only problem I am having, is that it adding a single “0” somewhere inside the dmx stream at a random position, shifting other values to the next channel this is happening at random times. Sometimes multiple times a second, sometimes it stays quiet for a couple of seconds.
Here is a output (with timestamp) of a readout of 20 channels, I’m using a Arya DMX controller which I configured to add 10 to every channel (so you can see the shift more clearly):

```
15:54:37.596 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 0, 140, 150, 160, 170, 180, 190, 
15:54:37.701 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:39.186 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 0, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:39.256 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:41.587 -> 10, 20, 30, 40, 50, 60, 70, 0, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:41.657 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:43.299 -> 10, 20, 30, 40, 0, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:43.367 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:46.707 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 0, 
15:54:46.809 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:49.406 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 0, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:49.474 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:49.715 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 0, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:49.782 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:54.426 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 0, 130, 140, 150, 160, 170, 180, 190, 
15:54:54.495 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:54:55.995 -> 10, 20, 30, 40, 50, 60, 70, 0, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:54:56.063 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:06.447 -> 10, 20, 30, 40, 50, 60, 0, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:06.516 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:07.545 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 0, 160, 170, 180, 190, 
15:55:07.613 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:08.881 -> 10, 20, 30, 40, 50, 60, 70, 0, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:08.983 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:10.076 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 0, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:10.177 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:14.580 -> 10, 20, 30, 40, 50, 60, 0, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:14.648 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:16.055 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 0, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:16.124 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:16.737 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 0, 160, 170, 180, 190, 
15:55:16.805 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:19.271 -> 10, 20, 30, 40, 50, 60, 0, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 
15:55:19.337 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200, 
15:55:23.969 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 0, 180, 190, 
15:55:24.038 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 0, 140, 150, 160, 170, 180, 190, 
15:55:24.108 -> 10, 20, 30, 40, 50, 60, 70, 80, 90, 100, 110, 120, 130, 140, 150, 160, 170, 180, 190, 200,  
```
As you can see there’s an extra 0 which shifts the values to the next channel. I’m thinking of doing a dirty fix by checking if the neighbouring channel has inherited the value and if so, ignoring the complete DMX stream. But I’m hoping there might be a neater solution.

luksal, 2020-03-03:
Can you provide a trace made with a logic analyzer to see if the electrical side works correct?

macdroid53, 2020-03-03:
It is probably the same issue I dealt with.

Once you get to 480 bytes read, a buffer read is not triggered because the the remaining, incoming don't trigger it. Then it sits until a break is detected. The break detect happens and the next byte is read into the FIFO before the interrupt triggers. If the buffer is not flushed there is actually 33 bytes (again, probably interrupt latency).

In https://github.com/espressif/esp-idf/issues/4537 I added the low level change to uart.c that would trigger a read event when the break is detected. Then I had to take special care in the higher level code to read the remaining 32 bytes, but, leave the 0 (which is the DMX start code).

luksal, 2020-03-03:
@macdroid53 I don't think that this is the problem

satoer, 2020-03-03:
I've put on a logic analyser, on the dmx side and the serial side. Both output the same data. I recorded couple of seconds when the anomaly happens. Analysed it as DMX data, and the data stays exactly the same (no anomaly in the stream). One thing I noticed. The DMX mixer only outputs 192 values / channels, maybe that's the problem?

Here's a piece of dmx data stream which constantly repeats (with another start code):
```
0.762522000000000,10,20,30,40,50,60,70,80,90,100,110,120,130,140,150,160,170,180,190,200,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,101,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
0.837180000000000,10,20,30,40,50,60,70,80,90,100,110,120,130,140,150,160,170,180,190,200,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,101,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
```
This is the end and start of the stream:
```
0.106832750000000,DMX-512,Slot 191: '0'
0.106864500000000,DMX-512,stop bits
0.106872750000000,DMX-512,MARK Time after Slot
0.107244750000000,DMX-512,start bit
0.107248750000000,DMX-512,Slot 192: '0'
0.107280500000000,DMX-512,stop bits
0.107288750000000,DMX-512,MARK Time after Slot
0.112497750000000,DMX-512,BREAK
0.112634000000000,DMX-512,Mark After Break
0.112642500000000,DMX-512,start bit
0.112646500000000,DMX-512,START CODE: '0'
0.112678250000000,DMX-512,stop bits
0.112686500000000,DMX-512,MARK Time after Slot
0.113030250000000,DMX-512,start bit
0.113034250000000,DMX-512,Slot 1: \n
0.113066000000000,DMX-512,stop bits
0.113074250000000,DMX-512,MARK Time after Slot
0.113446250000000,DMX-512,start bit
0.113450250000000,DMX-512,Slot 2: '20'
0.113482000000000,DMX-512,stop bits
0.113490250000000,DMX-512,MARK Time after Slot
```

satoer, 2020-03-05:
Oh crap. I'm so sorry. Ive been experimenting with so many libraries, I've mixed up the github repositories! I used the LXESP32DMX library. Feel so stupid right now. I'm really sorry for your time if you looked into the problem!

closed 2020-03-05 by satoer

luksal, 2020-03-05:
Just for my records, why do you think, that the LXESP32DMX is a better choice, even with the extra 0 problem?
As far as I have used this library the only issue is that you can't receive more that 480 channels, but this is a bug in the IDF framework and will hopefully fixed by the expressif developers.

satoer, 2020-03-10:
Sorry for the late reply. Actually this library seems to work great, and solved my problem with the extra “0”. But at the time when I was trying out several library’s (hence the library mix up) I could not get your library to work at all. (lots of error messages). I posted the "extra 0" problem in the right repository, but didn’t got any feedback how to solve this issue. So I tried to discover why I got these error messages in your library. Turns out adding:
```#include "arduino.h"```
At the source fixed the problem.
Don’t know why this is necessary but probably not needed in newer Arduino versions otherwise the problem probably would be mentioned before. (I use an older 1.8.9 version to be compatible with other library’s). 

So, for DMX receiving this library seems to work great. Sadly if I ever need to send DMX, it’s not possible with this library. So My preferred choice was to get the other library to work flawlessly.

If I could make a simple suggestion though, it would be great if these variables wouldn’t be baked in the library, but could be set in the user code:
```
#define DMX_SERIAL_INPUT_PIN    16          // pin for dmx rx
#define DMX_SERIAL_OUTPUT_PIN   17          // pin for dmx tx
```
If I have multiple project al always need to use the same pins, or adjust the library code.

Also, is it necessary to define an output pin for a read only library? (saves a I/O pin if not)

Anyway, thanks for the library!

luksal, 2020-03-10:
`#include "arduino.h"` is a know bug, that sometimes occurs, but I couldn't find out when and why it's sometimes necessary and sometimes not. See #11 for more information.

The library addresses the UART2 which is internal hardwired to pins 16 and 17, so there is no chance to use other pins. You can change the library to use UART1 (don't use UART0, because you can break the communication to your Arduino board) which uses two different pins, but I wouldn't recommend this.
Yes you could change it via the routing matrix, but this adds latency and in my testings I got some timing issues.

The output pin is reserved, because I'm working on a way to send DMX, see #3

satoer, 2020-03-10:
> The library addresses the UART2 which is internal hardwired to pins 16 and 17, so there is no chance to use other pins.

Hmmm This is my setup:
```
#define DMX_SERIAL_INPUT_PIN    33          // pin for dmx rx
#define DMX_SERIAL_OUTPUT_PIN   25          // pin for dmx tx
```
And it works flawlessly. I did not encounter any issue's. I'm using 14 channels, and the start channel is user settable between 1 and 256.  All the other pins are wired to other peripherals (Lights and dip switches). 

> #include "arduino.h" is a know bug, that sometimes occurs, but I couldn't find out when and why it's sometimes necessary and sometimes not.

Maybe if the user has the (popular) Fastled library installed this problem occurs. This Fastled library has also a ``dmx.h`` file, and I saw some mentions about this in the console window when it would not compile.  If I added the ``#include "arduino.h`` it compiled without errors. But I didn't investigate any further why this worked. 
> The output pin is reserved, because I'm working on a way to send DMX, see #3
Great!