luksal/ESP32-DMX #15: DMX sometimes return 0 value when Wifi is active

rickcard82, 2021-05-03:
Hello,
I'm developing a motor driver board that use DMX as input for positioning. The DMX task is running on the Core 0 and the motor task (with FastAccelStepper Lib) is running on the Core 1. Also I can enable the Wifi AP mode with a http server, both running on Core 0, to make some configuration about the board (set DMX Address and others board option).
When I enable the Wifi AP (by pressing a button), sometimes (about 1 or 2 times every 10 seconds) the DMX incoming data return 0 values for a frame about, and after the correct values comes back.
The DMX task priority is set to 3 and the Wifi task to 2, so I think the DMX has always the high priority. I tried to move the DMX task to Core 1 to make some tests, and the issue disappear, so I think it can be a problem relative to the Core (0) sharing between the two task. I'm using Arduino framework.
Also if the Wifi task hold busy the Core while DMX data is coming to the UART, should RX data be safe by the UART RX ring buffer? I'd like to try to change some parameters about the full FIFO threshold but this is supported only by using the esp-idf framework. Any suggestion? WiFi starts other "system" task with higher priority ? I'v tried to set the DMX task priority to the highest value but nothing changes.
Thanks
R

luksal, 2021-05-04:
Unfortunately I can give no support for the Arduino Framework. But I recommend to use 1 core exclusive (or very unimportant tasks) for all Wifi and network communication.
Maybe it's a problem with the internal interrupt architecture of the esp32, but I don't know how it works. From my own experiments I can say, the esp32 is easy to use for simple tasks, but if you need a reliable controller, choose something else and don't use the Arduino framework...

rickcard82, 2021-05-04:
Thanks for your reply. When you talk about reliable controller it's a quality reason about esp32? I always used microchip and texas instruments micro, and now I tried to use the esp32 after I saw it's used for a lot of projects and it's is very cheap for it's "skills" ( mean 2 cores at 240MHz and a lot or peripherals). So if I understand right your lib is develop for esp idf or it is a porting for the esp32? I don't understand how you developed it if you don't know about esp32 architecture :) about arduino framework  I think to migrate to esp idf framework (I have to study it), by some forums and sites all developers advise it.

luksal, 2021-05-06:
It may be a great controller, but the ESP-IDF is a mess and the Arduino port, which builds on top of the Ida ist not better...
The specs are okisch, but if you look into detail, there are a lot of compromises they made to place all this into a single die.
And yes, because it's full packed, you can get a lot of trouble...
I would recommend to stick with microchip or ti. Or switch to ST ;) 
This library builds on top of the ESP-IDF, and I have a little bit of knowledge about the UART peripheral in the ESP32 but nothing more. And yes, there a bugs inside of the IDF which made the complete spec compliant DMX implementation nearly impossible... See #1 for details

luksal, 2021-05-06:
Closing issue, because it's not library relevant and seems to be inside ESP-IDF

closed 2021-05-06 by luksal