luksal/ESP32-DMX #18: Compile error

peerke444, 2021-09-04:
I'm trying to compile the SimpleRX.ino example but keep getting an error.
Does anybody know what I'm doing wrong?

This is the error:
```
/Users/peter/Dropbox/Arduino/MySketches/libraries/ESP32-DMX-master/examples/SimpleRX/SimpleRX.ino: In function 'void setup()':
SimpleRX:7:19: error: no matching function for call to 'DMX::Initialize()'
   DMX::Initialize();
                   ^
In file included from /Users/peter/Dropbox/Arduino/MySketches/libraries/ESP32-DMX-master/examples/SimpleRX/SimpleRX.ino:1:0:
/Users/peter/Dropbox/Arduino/MySketches/libraries/ESP32-DMX-master/src/dmx.h:35:21: note: candidate: static void DMX::Initialize(DMXDirection)
         static void Initialize(DMXDirection direction);     // initialize library
                     ^
/Users/peter/Dropbox/Arduino/MySketches/libraries/ESP32-DMX-master/src/dmx.h:35:21: note:   candidate expects 1 argument, 0 provided
Meerdere bibliotheken gevonden voor "dmx.h"
Gebruikt: /Users/peter/Dropbox/Arduino/MySketches/libraries/ESP32-DMX-master
Niet gebruikt: /Users/peter/Dropbox/Arduino/MySketches/libraries/FastLED
exit status 1
no matching function for call to 'DMX::Initialize()'
```

marmilicious, 2021-09-04:
Error says "candidate expects 1 argument, 0 provided"
Try putting "input" in there?
DMX::Initialize(input)

luksal, 2021-09-05:
@marmilicious is right.

I recently added code to be able to send dmx with the library as well. Unfortunately I haven't had time to adapt the old DMX receiving example, let alone add a new example for sending.

With the in the example missing parameter `input` or `output`, it is possible to specify whether to receive or transmit when initializing the class.

So to receive DMX change the `DMX::Initialize()` to `DMX::Initialize(input)`

peerke444, 2021-09-05:
Yes you're right @marmilicious.
The sketch is compiling now.

Unfortunately I can't receive any DMX yet but I'm afraid thats a hardware issue.
Always something to do next. :-) 

Thanks @marmilicious and @luksal for the help!

closed 2021-09-05 by peerke444